keyboardio/Model01-Firmware #8: Releasing `fn` before key enters 2 characters

bergey, 2017-07-28:
On the PVT with default firmware, on Windows 10, releasing the `fn` key while still holding one of `uiop` prints both characters - `{u`, `[o`- and so forth.  I find I roll my hand in a way that I do this frequently.  Does anyone want this behavior?

obra, 2017-07-28:
>    Reverse link: [3]unknown
> References
>    Visible links

I think I agree that it's not desirable. On Windows 10, does shift
behave the same way?

obra, 2017-07-28:
Because this is a more general Kaleidoscope issue, I've created keyboardio/Kaleidoscope#150

obra, 2017-07-28:
I'm closing this issue, but that doesn't mean the issue is resolved -- just that I've moved the issue to the correct bugtracker. (Not that you did anything wrong by opening it here, @bergey )

closed 2017-07-28 by obra