keyboardio/Model01-Firmware #11: /dev/ttyACM* doesn't exist, can't flash

wittrock, 2017-08-07:
While trying to change up my layout, I ran across

```sh
~/kaleidoscope/Model01-Firmware(master) » make flash                                                                                                         jwittrock@john
/Users/jwittrock/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope-Plugin/tools//kaleidoscope-builder flash
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.02-15-g60ed) ...
Press ENTER when ready...

stty: 1200: No such file or directory
make: *** [flash] Error 1```

Typing works normally, it's just flashing that fails.

I dug around a bit, and found this in `Arduino-Boards/libraries/Kaleidoscope-Plugin/tools/kaleidoscope-builder.conf`:

```sh
RESET_DEVICE="stty -F ${DEVICE_PORT} 1200 hupcl"
```

While the keyboard is plugged in, I don't see any `ttyACM*` devices in /dev, but it seems like this is required. Any idea how I can restore that entry?

algernon, 2017-08-07:
Are you on Linux, or OSX? If you are on Linux, can you unplug your keyboard, run `dmesg -Tw`, plug it in, and paste the output?

wittrock, 2017-08-07:
I'm on OSX, but on Linux the flashing either gives me a `Permission denied` or takes > 30s to connect to the programmer (so I aborted it). Those seem like a different issue though :)

Is there an OSX equivalent for the debug output you're looking for? I'm guessing there has to be a way to get similar USB enumeration output, but I'm not sure what it is...

wittrock, 2017-08-07:
I've got the flashing working on Linux by setting the correct group for my user and completely purging `modem-manager` from the system (it seems to have been holding `/dev/ttyACM0` open, preventing flashing).

Still no luck on Mac though, the file doesn't even appear. Still looking into options for Mac USB device enumeration.

algernon, 2017-08-08:
On mac, you need some other device, not `/dev/ttyACM0`. It is one of `/dev/cu.usbmodemkbio*`, `/dev/cu.usbmodemHID*`, or `/dev/cu.usbmodem14*` as far as I can tell.

For linux, if you install the [udev rules](https://github.com/keyboardio/Kaleidoscope/blob/master/etc/99-kaleidoscope.rules), that should prevent modem-manager from jumping on the serial port.

cblake777, 2017-08-10:
I'm also have the same issue.  Oddly, however, I was able to successfully flash the keyboard once.  After that successful flash, it started giving the `stty` error noted above when I attempted to run it again.  

As far as devices go, the only device I have that is similar to the ones you've noted is `/dev/cu.usbmodemCHIDHL1`.

obra, 2017-08-10:
>    Reverse link: [3]unknown
> References
>    Visible links

Inside your checkout of the Arduino-Boards repo, can you try:

make update-submodules

and then try to flash again? I've updated the wildcards for the keyboard
paths we search for.

cblake777, 2017-08-10:
Woo! I now have F14/F15 (brightness controls) mapped :D

It works!  That did the trick.

Thanks.

obra, 2017-08-10:
Awesome! I'm happy to hear it. @wittrock, can you give this a try? If it doesn't behave, please feel free to reopen this ticket.

closed 2017-08-10 by obra

pian0, 2017-11-06:
I'm seeing a similar issue on Mac OS. Looking in my /dev/ shows I have a device cu.usbmodem61. After unplugging and replugging the model 01 a couple of times I saw 62 and 63 also appear.

obra, 2017-11-06:
@pian0 - That _sounds_ like you might have used an older Arduino IDE
version to flash the keyboard, which would cause our USB Port naming stuff
to get overriden by Arduino's. But it might also be tied to Arduino
version. Can you tell me what version of the Arduino IDE you're using and
what version of OSX?

pian0, 2017-11-06:
I only started hacking on this last Friday so I don't think the version should be out of date.

Arduino 1.8.5
macOS 10.12.6

Adding this line in /Users/ian/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/etc/kaleidoscope-builder.conf
allowed me to flash the firmware once:
`DEVICE_PORT="${DEVICE_PORT:-$(ls /dev/cu.usbmodem63* 2> /dev/null || echo '')}"`
Now the device is 64 in /dev/ so it seems like replugging the device or flashing its firmware increments the number.

obra, 2017-11-06:
That's not _supposed_ to be the case anymore.  You should be seeing a port
name of /dev/cu.usbmodem*kbio*, but clearly something is up. Did you
install our Arduino support by cloning Arduino-Boards or by using the
Arduino.app boards manager?

pian0, 2017-11-06:
I cloned Arduino-Boards.

I see `/dev/cu.usbmodemCCkbio01` and after restarting my machine it no longer creates the incrementing device name.

obra, 2017-11-06:
Innnteresting. I'm glad it sounds like it's started to behave, though I
appreciate the report. It sounds like we've still got some work there..