keyboardio/Model01-Firmware #29: Numlock numbers don't work

gsamuelhays, 2017-10-30:
After hitting the num key, the numberpad area lights up on the keyboard but numbers don't actually work. Some symbols seem to.

Confirmed on Windows 7 and Fedora Linux. I have not reflashed the firmware, this is OOB keyboardio Model 01 with default firmware.

algernon, 2017-10-30:
I think the root cause of this is the same as in keyboardio/Kaleidoscope-NumLock#10 and keyboardio/Kaleidoscope-NumLock#11. The good news is that work is under way to fix these properly!

warewolf, 2018-01-28:
Got a bugreport for Keyboardio Model 01 - Kaleidoscope sketch:v1.22;lib:v1.22.

This is going to be a little confusing.  Know how other keyboards have a set of 3 leds for num/caps/scroll lock?  It looks like somehow the Model 01 is *enabling* the "numlock is enabled" LED, but not *disabling* it when the keyboard comes out of numlock mode.

Wild guess here, based from observation with xev(1):  The keyboard sees/knows the status of the "numlock is enabled" LED and sends numlock keydown/keyup events going *in* numlock mode, but does not send keydown/keyup events going *out* of numlock mode.

How'd I find this?  It broke my FVWM root menus from the mouse.  FVWM treats the numlock as a modifier and shows different (unexpected) menus.