keyboardio/Model01-Firmware #33: not possible to combine alt & shift in stock firmware

boqs, 2017-11-03:
I can't combine the alt & shift modifiers, which is important for some of the 'stock' emacs bindings.

- chording "alt shift ." has the same effect as "alt ."
- chording "shift alt ." it has the same effect as "shift ."

I guess this won't be a hard fix, because ctrl & shift *do* seem to play nice together.  For example the default emacs binding for undo is the chord "ctrl shift -".  This works as expected.

So is there some firmware difference between alt & ctrl?

algernon, 2017-11-03:
Which of the Alt keys are you trying to use? Most things don't make a difference between left- or right-control or shift, but right-alt is special (also called `AltGr`). If you are using that, perhaps that's what's tripping Emacs up?

I just tried with OneShots, and `Alt+Shift+.` got through as `M->`, as expected. Will try with the stock firmware too, but if it works with OneShots, it should work without too.

boqs, 2017-11-03:
oops - sorry for the line noise.  I think this bug is something to do with my weird system setup & the fact that I'm usually remapping AltGr to Alt through X windows.

I'll close this issue until I've eliminated that from the equation

closed 2017-11-03 by boqs