keyboardio/Model01-Firmware #37: No longer able to build the firmware after doing a `git pull` just now

svanharmelen, 2017-11-14:
I was able to `make flash` without issues a couple of times, but after I just did a `git pull` it now breaks with the following error:

```
BOARD_HARDWARE_PATH="/Users/svanharmelen/Documents/Arduino/hardware" /Users/svanharmelen/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22-dirty) ...
/Users/svanharmelen/Documents/Arduino/Model01-Firmware/Model01-Firmware.ino:29:33: fatal error: Kaleidoscope-NumPad.h: No such file or directory
 #include "Kaleidoscope-NumPad.h"
                                 ^
compilation terminated.
exit status 1
make: *** [flash] Error 1
```

svanharmelen, 2017-11-14:
FYI: I also did a `git pull` in the `hardware/keyboardio/avr` folder/repo, but that didn't solve anything...

obra, 2017-11-14:
In the Arduino-Boards repo, can you "make update-submodules"?

svanharmelen, 2017-11-14:
Works like a charm 👍  Thx!

closed 2017-11-14 by svanharmelen