keyboardio/Model01-Firmware #38: Function key interaction degrades mouse usage on Mac OS X

abiyogi, 2017-11-29:
Using the function key after a few minutes creates an issue with mouse/trackpad on Mac OS X. Unplugging and plugging back the keyboard fixes this, after another 20-30 minutes, the same happens. To reproduce this issue, simply use Model 01 on a Mac and use the function keys a few times and try right clicking with your mouse/trackpad.

algernon, 2017-11-29:
What issue are you seeing? The clicks getting stuck, perhaps?

abiyogi, 2017-11-29:
Yes, the clicks getting stuck. Both my Magic Mouse and trackpad loose right click and also some visual bugs. There seems to be a bug in the function key toggle w.r.t. the keyboard and it is triggering some signals/key-bouncing issue which makes the mouse misbehave. Is there a way for me to provide some logs? Happy to help!

algernon, 2017-11-29:
No need to, this is a [known issue](https://github.com/keyboardio/Kaleidoscope-MouseKeys/issues/10), and a fix is pending. If you want to give it a go, and are comfortable with git, you can try keyboardio/Kaleidoscope-MouseKeys#12 (and the two other PRs it depends on).

The root cause is that if you release `Fn` sooner than the mouse key, the mouse key will never get released, and will appear stuck. You can fix it by pressing `Fn` and the mousekey again, and releasing the mousekey first.

The referenced PR addresses this, and will allow you to release keys in any order.

abiyogi, 2017-11-29:
Thanks, I should have looked into the other repository too! I'll close this one.

algernon, 2017-11-29:
Starting in this repo is good practice. It is sometimes hard to tell which plugin is responsible for a bug, so filing one against `Model01-Firmware` is a safe bet. It is not a big deal if its a duplicate. :)

abiyogi, 2017-11-29:
Alright, and yes I am familiar with git, I'll test the commits once I get some time.

gedankenexperimenter, 2017-12-12:
This issue can probably be closed now, can't it?

algernon, 2017-12-12:
Yep, this seems to be in Arduino-Boards too. Thanks for the reminder!

closed 2017-12-12 by algernon