keyboardio/Model01-Firmware #40: fatal error: Kaleidoscope-NumPad.h: No such file or directory  #include "Kaleidoscope-NumPad.h"

benshine, 2017-12-04:
I can't tell if this is just my setup (which I'm struggling with) or a legit error. 
```
ben-15:Model01-Firmware ben$ pwd
/Users/ben/kaleidoscope/Model01-Firmware
ben-15:Model01-Firmware ben$ make flash
BOARD_HARDWARE_PATH="/Users/ben/Library/Arduino15/packages/keyboardio/hardware/avr/1.19.0" /Users/ben/Library/Arduino15/packages/keyboardio/hardware/avr/1.19.0/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22) ...
/Users/ben/kaleidoscope/Model01-Firmware/Model01-Firmware.ino:29:33: fatal error: Kaleidoscope-NumPad.h: No such file or directory
 #include "Kaleidoscope-NumPad.h"
                                 ^
compilation terminated.
exit status 1
make: *** [flash] Error 1
```

That file does exist, though:
```
ben-15:Model01-Firmware ben$ cd ..
ben-15:kaleidoscope ben$ ls
Arduino-Boards		Kaleidoscope		Model01-Firmware
ben-15:kaleidoscope ben$ find . -name Kaleidoscope-NumPad.h
./Arduino-Boards/libraries/Kaleidoscope-NumPad/src/Kaleidoscope-NumPad.h
```

This does seem like a problem with my installation -- what am I missing here?

benshine, 2017-12-04:
After I removed all references to NumPad from the ino, I was able to flash my keyboard successfully.

obra, 2017-12-04:
I suspect that perhaps your 'Arduino-Boards' isn't update or you haven't
run 'make update-submodules' inside it recently enough.

benshine, 2017-12-04:
Hmm, nope, they're up to date:
```
cd ../Arduino-Boards/
ben-15:Arduino-Boards ben$ make update-submodules
git pull
Already up to date.
git submodule update --init --recursive
All Kaleidoscope libraries have been updated from GitHub
```
& note that the `Kaleidoscope-NumPad.h` file did exist already.

obra, 2017-12-04:
Ok. Next up: Do you have this directory: ~/.arduino15/packages/keyboardio/ ?

If so, you might have installed an older version of the libraries with the
Arduino board manager.

benshine, 2017-12-04:
nope, but i do often launch arduino ui, which has a pretty amazing preferences.txt file
```
board=model01
boardsmanager.additional.urls=https://raw.githubusercontent.com/keyboardio/boardsmanager/master/package_keyboardio_index.json
build.verbose=false
build.warn_data_percentage=75
compiler.cache_core=true
compiler.warning_level=none
console=true
console.auto_clear=true
console.error.file=stderr.txt
console.length=500
console.lines=4
console.output.file=stdout.txt
editor.antialias=true
editor.auto_close_braces=true
editor.caret.blink=true
editor.code_folding=true
editor.divider.size=0
editor.external=false
editor.font=Monaco,plain,12
editor.indent=true
editor.invalid=false
editor.keys.alternative_cut_copy_paste=false
editor.keys.home_and_end_beginning_end_of_doc=false
editor.keys.shift_backspace_is_delete=true
editor.languages.current=
editor.linenumbers=true
editor.save_on_verify=true
editor.tabs.expand=true
editor.tabs.size=2
editor.update_extension=true
editor.window.height.default=600
editor.window.height.min=290
editor.window.width.default=500
editor.window.width.min=400
export.applet.separate_jar_files=false
export.application.fullscreen=false
export.application.platform=true
export.application.stop=true
export.delete_target_folder=true
gui.scale=auto
last.folder=/Users/ben/src/Model01-sketch/src/algernon/algernon.ino
last.ide.1.8.5.daterun=1512364400
last.ide.1.8.5.hardwarepath=/Applications/Arduino.app/Contents/Java/hardware
last.screen.height=1440
last.screen.width=2560
last.sketch.count=1
last.sketch.default.location=110,168,1002,1148,623,0
last.sketch.default.path=/Users/ben/src/model01/Model01-programmer-dvorak/Model01-programmer-dvorak.ino
last.sketch0.location=110,168,1002,1148,623,0
last.sketch0.path=/Users/ben/src/model01/Model01-programmer-dvorak/Model01-programmer-dvorak.ino
last.sketch1.location=110,168,1002,1148,623,0
last.sketch1.path=/Users/ben/src/model01/Model01-programmer-dvorak/Model01-programmer-dvorak.ino
last.sketch2.location=110,168,1002,1148,623,0
last.sketch2.path=/Users/ben/src/model01/Model01-programmer-dvorak/Model01-programmer-dvorak.ino
last.sketch3.location=419,213,1214,1174,686,0
last.sketch3.path=/Users/ben/src/model01/Model01-programmer-dvorak/Model01-programmer-dvorak.ino
platform.auto_file_type_associations=true
preproc.color_datatype=true
preproc.enhanced_casting=true
preproc.imports.list=java.applet.*,java.awt.Dimension,java.awt.Frame,java.awt.event.MouseEvent,java.awt.event.KeyEvent,java.awt.event.FocusEvent,java.awt.Image,java.io.*,java.net.*,java.text.*,java.util.*,java.util.zip.*,java.util.regex.*
preproc.output_parse_tree=false
preproc.save_build_files=false
preproc.substitute_floats=true
preproc.substitute_unicode=true
preproc.web_colors=true
programmer=arduino:avrispmkii
proxy.manual.hostname=
proxy.manual.password=
proxy.manual.port=
proxy.manual.type=HTTP
proxy.manual.username=
proxy.pac.url=
proxy.type=auto
recent.sketches=/Users/ben/Documents/Arduino/hardware/keyboardio/avr/libraries/Model01-Firmware/Model01-Firmware.ino,/Users/ben/Library/Arduino15/packages/keyboardio/hardware/avr/1.19.0/libraries/Model01-Firmware/Model01-Firmware.ino,/Users/ben/src/model01/Model01-Firmware/Model01-Firmware.ino,/Users/ben/src/Model01-sketch/src/algernon/algernon.ino,/Users/ben/src/model01/Model01-programmer-dvorak/Model01-programmer-dvorak.ino,/Users/ben/src/model01/Model01-programmer-dvorak-simplify/Model01-programmer-dvorak-simplify.ino,/Users/ben/src/model01/Model01-Firmware-experiments/Model01-Firmware-experiments.ino
run.display=1
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=true
run.present.stop.color=#cccccc
serial.databits=8
serial.debug_rate=9600
serial.parity=N
serial.port=/dev/cu.usbmodemCkbio01
serial.port.file=cu.usbmodemCkbio01
serial.port.iserial=null
serial.stopbits=1
sketchbook.path=/Users/ben/Documents/Arduino
software=ARDUINO
target_package=keyboardio
target_platform=avr
update.check=true
update.id=8769879744087831657
upload.using=bootloader
upload.verbose=true
upload.verify=true

```

Am I causing myself trouble by using both the gui and the command line? I have to fall back to the gui to re-install my good layout after I destroy things with the command line :)

obra, 2017-12-04:
Oh. OSX has the file I was looking for in another place:

/Users/ben/Library/Arduino15/packages/keyboardio/hardware/avr/1.19.0

You should _either_ use Arduino-Boards from git or the Keyboardio
support from the 'Boards Manager' in Arduino.

Right now, you've got both.

It's safe to mix the cli and the gui.

benshine, 2017-12-04:
Aha! I do have those files! I've removed the boards manager from the gui, and rm -r'd `/Users/ben/Library/Arduino15/packages/keyboardio/hardware/avr/1.19.0` 
but still no joy. I'll move this to the forums, because this sounds like a problem with my configuration/install, rather than a problem in the repo.

closed 2017-12-04 by benshine