keyboardio/Model01-Firmware #44: make flash error on macOS

masukomi, 2018-03-01:
after following setup instructions in README

Keyboard was plugged in. Confirmed by lit LEDs and having typed on it.

After hitting ENTER lights went out.

```
make flash
BOARD_HARDWARE_PATH="/Users/masukomi/Documents/Arduino/hardware" /Users/masukomi/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22-3-gb745) ...
Press ENTER when ready...

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = "��"; type = w
    Software Version = .; Hardware Version = .
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = "��"; type = G
    Software Version = .; Hardware Version = �.
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
```

seemed to continue on in this vein. Tried to ^C out of it but that didn't work.

Unplugging keyboard did kill it.

obra, 2018-03-01:
Hi,

Did you hold down the 'Prog' key while trying to flash?

masukomi, 2018-03-02:
I did not. I was waiting for the prior instruction to complete. 

Might it be better to change the command line instructions to be what they actually need to be rather than half of what they need to be? That is to say 

`Press ENTER when ready`

should be

`Hold down prog and press ENTER when ready`

Possibly even "and continue holding down prog until ..."

masukomi, 2018-03-02:
and thank you @obra that did address the problem.

algernon, 2018-03-05:
Yep, changing the line printed would be great. Do you want to make a pull request that does so? :)

Gastove, 2018-03-10:
@algernon I was poking at Kaleidoscope anyways, and it turns out the instruction comes from there. [Voila](https://github.com/keyboardio/Kaleidoscope/pull/306).

gedankenexperimenter, 2019-03-07:
I believe this issue has been fully addressed in [39601fc](https://github.com/keyboardio/Kaleidoscope/commit/39601fcba2d7b9bdf0a98aafd26aa88505ef9330).

algernon, 2019-03-09:
Indeed, it is. Thanks for digging up the fix!

closed 2019-03-09 by algernon