keyboardio/Model01-Firmware #51: Multiple characters on keypress

jonathanasdf, 2018-05-14:
Not sure if this is firmware or something else. My 'a' key likes to randomly print 2-3 extra as sometimes, forcing me to go back and delete the extra ones. Additionally, sometimes after pressing the key the input is delayed a bit so when typing fast the letter can end up interchanged with the next one eg. asdf -> sadf.

This only happens with the 'a' key and no other keys.

noseglasses, 2018-05-14:
You might be experiencing key chatter. See [here](https://community.keyboard.io/t/what-to-do-if-your-keyboard-seems-to-duplicate-keystrokes-or-a-key-stops-working/636) and [here](https://community.keyboard.io/t/what-to-do-if-you-experience-key-chatter/959) for discussion about it and Jesse talking about a solution.

obra, 2018-05-14:
shoot us mail at help@keyboard.io and we can get you sorted out.