keyboardio/Model01-Firmware #54: make flash doesn't work

wt, 2018-06-01:
I am sync'd to the latest master (c7fab58f63b99013baa89b6637f36ad7be3bb3bc).
It will not build. Here's some output to help:

```
$ make flash
BOARD_HARDWARE_PATH="/home/wt/Arduino/hardware" /home/wt/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building output/Model01-Firmware/Model01-Firmware (0.0.0-gv1.22-13-gc7fa) ...
/home/wt/Arduino/Model01-Firmware/Model01-Firmware.ino:285:26: error: expected constructor, destructor, or type conversion before '(' token
 KALEIDOSCOPE_INIT_PLUGINS(
                          ^
exit status 1
/home/wt/Arduino/hardware/keyboardio/avr/build-tools/makefiles//rules.mk:72: recipe for target 'flash' failed
make: *** [flash] Error 1
```

obra, 2018-06-02:
Hi Wren,

Are you also current on the Arduino-Boards repo? Once you pull that one,
you also need to 'make update-submodules' inside it.

Best,
Jesse

obra, 2018-08-11:
I'm closing this for now due to the lack of a reply. Feel free to reopen it if you still need a hand, @wt.

closed 2018-08-11 by obra