keyboardio/Model01-Firmware #60: Stuck in numlock on modified keymap

rknLA, 2018-07-26:
I've modified my keymap in the firmware so that the `any` key activates numlock instead of the `num` key, but when I enter Numlock mode, the fall-through for the `any` key continues to send the macro-version-info text out the pipe.  As such, the only way for me to leave numlock mode is to unplug and re-plug-in my keyboard.

Anyone have any ideas about how to fix this?

I'm on Mac OS 10.13.4, if that makes any difference, and my customized keymap looks like this:

```
KEYMAPS(

  [QWERTY] = KEYMAP_STACKED
  (Key_Escape,   Key_1, Key_2, Key_3, Key_4, Key_5, Key_LEDEffectNext,
   Key_Tab, Key_Q, Key_W, Key_E, Key_R, Key_T, Key_Tab,
   Key_LeftControl,   Key_A, Key_S, Key_D, Key_F, Key_G,
   Key_LeftShift, Key_Z, Key_X, Key_C, Key_V, Key_B, Key_LeftGui,
   Key_Backtick, Key_Spacebar, Key_Enter, Key_LeftAlt,
   ShiftToLayer(FUNCTION),

   LockLayer(NUMPAD),  Key_6, Key_7, Key_8,     Key_9,         Key_0,         Key_Minus,
   Key_Enter,     Key_Y, Key_U, Key_I,     Key_O,         Key_P,         Key_Equals,
                  Key_H, Key_J, Key_K,     Key_L,         Key_Semicolon, Key_Quote,
   Key_RightAlt,  Key_N, Key_M, Key_Comma, Key_Period,    Key_Slash,     Key_RightShift,
   Key_RightShift, Key_LeftAlt, Key_Backspace, Key_Minus,
   ShiftToLayer(FUNCTION)),

  [NUMPAD] =  KEYMAP_STACKED
  (___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___,
   ___,

   ___,                    ___, Key_Keypad7, Key_Keypad8,   Key_Keypad9,        Key_KeypadSubtract,  M(MACRO_VERSION_INFO),
   ___,                    ___, Key_Keypad4, Key_Keypad5,   Key_Keypad6,        Key_KeypadAdd,      ___,
                           ___, Key_Keypad1, Key_Keypad2,   Key_Keypad3,        Key_Equals,         ___,
   ___,                    ___, Key_Keypad0, Key_KeypadDot, Key_KeypadMultiply, Key_KeypadDivide,   Key_Enter,
   ___, ___, ___, ___,
   ___),

  [FUNCTION] =  KEYMAP_STACKED
  (___,      Key_F1,           Key_F2,      Key_F3,     Key_F4,        Key_F5,           XXX,
   Key_Tab,  ___,              Key_mouseUp, ___,        Key_mouseBtnR, Key_mouseWarpEnd, Key_mouseWarpNE,
   Key_Home, Key_mouseL,       Key_mouseDn, Key_mouseR, Key_mouseBtnL, Key_mouseWarpNW,
   Key_End,  Key_PrintScreen,  Key_Insert,  ___,        Key_mouseBtnM, Key_mouseWarpSW,  Key_mouseWarpSE,
   ___, Key_Enter , ___, ___,
   ___,

   Consumer_ScanPreviousTrack, Key_F6,                 Key_F7,                   Key_F8,                   Key_F9,          Key_F10,          Key_F11,
   Consumer_PlaySlashPause,    Consumer_ScanNextTrack, Key_LeftCurlyBracket,     Key_RightCurlyBracket,    Key_LeftBracket, Key_RightBracket, Key_F12,
                               Key_LeftArrow,          Key_DownArrow,            Key_UpArrow,              Key_RightArrow,  ___,              ___,
   Key_PcApplication,          Consumer_Mute,          Consumer_VolumeDecrement, Consumer_VolumeIncrement, ___,             Key_Backslash,    Key_Pipe,
   ___, ___, Key_Delete, ___,
   ___)

	) // KEYMAPS(

```

obra, 2018-07-26:
you’ll want to remove the following from your numpad map

> On Jul 26, 2018, at 6:33 AM, R. Kevin Nelson <notifications@github.com> wrote:
> 
> M(MACRO_VERSION_INFO)

rknLA, 2018-07-27:
Thanks for the quick reply.  I've had a bit of a derp moment: I edited the Issue I filed in the old/wrong repo (https://github.com/keyboardio/Kaleidoscope-Numlock/issues/17)

As mentioned there, I have a workaround of specifying `LockLayer(NUMPAD)` in both layers, but I don't understand why the triple-underscore fallthrough isn't working.

Basically, I've gone from this:
```
  [NUMPAD] =  KEYMAP_STACKED
  (___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___,
   ___,

   ___,                    ___, Key_Keypad7, Key_Keypad8,   Key_Keypad9,        Key_KeypadSubtract,  M(MACRO_VERSION_INFO),
   ___,                    ___, Key_Keypad4, Key_Keypad5,   Key_Keypad6,        Key_KeypadAdd,      ___,
                           ___, Key_Keypad1, Key_Keypad2,   Key_Keypad3,        Key_Equals,         ___,
   ___,                    ___, Key_Keypad0, Key_KeypadDot, Key_KeypadMultiply, Key_KeypadDivide,   Key_Enter,
   ___, ___, ___, ___,
   ___),
```

to this:
```
  [NUMPAD] =  KEYMAP_STACKED
  (___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___, ___, ___, ___,
   ___, ___, ___, ___,
   ___,

   LockLayer(NUMPAD),                    ___, Key_Keypad7, Key_Keypad8,   Key_Keypad9,        Key_KeypadSubtract,  M(MACRO_VERSION_INFO),
   ___,                    ___, Key_Keypad4, Key_Keypad5,   Key_Keypad6,        Key_KeypadAdd,      ___,
                           ___, Key_Keypad1, Key_Keypad2,   Key_Keypad3,        Key_Equals,         ___,
   ___,                    ___, Key_Keypad0, Key_KeypadDot, Key_KeypadMultiply, Key_KeypadDivide,   Key_Enter,
   ___, ___, ___, ___,
   ___),
```

Removing the `M(MACRO_VERSION_INFO)` from the NUMPAD layer isn't what I want, as I would expect doing that means I'd hit the `any` key to enter numpad, and then have to hit the `num` key to leave it; I'd rather have the layer lock be on the same key to enter/exit.

obra, 2018-07-27:
Sorry, I appear to have misread your original bug report.

What I thought you said was that if you swap the position
of  LockLayer(NUMPAD) to your Any key, removing the M(ANY) macro as you do
so and then move M(MACRO_VERSION_INFO) on the numpad layer  to the
top-right corner, you're -still- having M(MACRO_VERSION_INFO) trigger when
you tap LockLayer(NUMPAD) for a second time.

Is that right?
ᐧ

rknLA, 2018-07-27:
Yes, pretty much.  The Any key is assigned to LockLayer(NUMPAD) instead of M(ANY), and the Any key in the NUMPAD layer is assigned to fall-through.  I'm assuming that this falls through to the other layer, and not the default-factory keymapping?

If I have LockLayer(NUMPAD) explicitly assigned to the Any key in both layers, then it works as-expected.

obra, 2018-07-27:
Can you attach your entire sketch?
ᐧ

rknLA, 2018-07-31:
Full sketch is up at https://gist.github.com/rknLA/a90c04a4e1544923110fb85af8f36dc1