keyboardio/Model01-Firmware #78: Magic combo or something to disable eeprom keymaps

obra, 2019-02-05:
Flashing the Model 01's firmware with a firmware that supports eeprom keymaps and having not-quite-perfect data in eeprom can lead to a situation where the keyboard appears to be 'bricked'

We need something like a magic combo to reset the eeprom enough to switch back to using the hardcoded keymaps. I think this is just resetting default layer and whether to hide the hardcoded keymaps

I think I actually want the key combo to only work if held in at boot. Let's call it "LeftFn-Cmd-1"

qayshp, 2019-02-05:
My keyboardio miiiiiiiiiiiiiiiiiight be in this condition.

qayshp, 2019-02-05:
I flashed it with the latest shiny stuff about 30 minutes ago and haven't been able to get it to work since.
Any suggestions on how to recover it?

algernon, 2019-02-05:
You can either flash an older firmware, or fire up Chrysalis, go to the layout editor, find the "..." menu, and turn off the "Use custom layers only" option. Both should work.

qayshp, 2019-02-05:
Yep, just got it recovered a minute ago after following the community thread and trying Chrysalis. It actually auto-detected the keyboard and said a firmware update was required.

All good for now! Sorry for spam on this issue :(