keyboardio/Model01-Firmware #90: Building fails?

mpalmujoki, 2019-09-18:
Hi,

I'm using Windows, with Arduino latest version.

I've followed the instructions, cloned the Kaleidescope into the hardware folder.
I also cloned this git repo.

I opened the Model01-Firmware.ino with Arduino.

However, verifying the build fails:

```
Invalid library found in C:\Users\mikpal\Documents\Arduino\libraries\Model01-Firmware-modded: no headers files (.h) found in C:\Users\mikpal\Documents\Arduino\libraries\Model01-Firmware-modded
Invalid library found in C:\Users\mikpal\Documents\Arduino\libraries\Model01-Jesses-Latest: no headers files (.h) found in C:\Users\mikpal\Documents\Arduino\libraries\Model01-Jesses-Latest
In file included from C:\Users\mikpal\Documents\Arduino\hardware\keyboardio\avr\libraries\Kaleidoscope\src/kaleidoscope/Kaleidoscope.h:60:0,

                 from C:\Users\mikpal\Documents\Arduino\hardware\keyboardio\avr\libraries\Kaleidoscope\src/Kaleidoscope.h:19,

                 from C:\Users\mikpal\Documents\Arduino\libraries\Model01-Firmware-modded\Model01-Firmware-1.22-modded\Model01-Firmware-1.22-modded.ino:17:

C:\Users\mikpal\Documents\Arduino\libraries\Model01-Firmware-modded\Model01-Firmware-1.22-modded\Model01-Firmware-1.22-modded.ino: In static member function 'static kaleidoscope::EventHandlerResult kaleidoscope::Hooks::onSetup()':

C:\Users\mikpal\Documents\Arduino\hardware\keyboardio\avr\libraries\Kaleidoscope\src/kaleidoscope_internal/event_dispatch.h:149:7: internal compiler error: Segmentation fault

       }                                                                   __NL__ \

       ^

C:\Users\mikpal\Documents\Arduino\hardware\keyboardio\avr\libraries\Kaleidoscope\src/kaleidoscope/event_handlers.h:98:4: note: in expansion of macro '_REGISTER_EVENT_HANDLER'

    OPERATION(onSetup,                                                     __NL__ \

    ^

C:\Users\mikpal\Documents\Arduino\hardware\keyboardio\avr\libraries\Kaleidoscope\src/kaleidoscope_internal/event_dispatch.h:201:3: note: in expansion of macro '_FOR_EACH_EVENT_HANDLER'

   _FOR_EACH_EVENT_HANDLER(_REGISTER_EVENT_HANDLER)                            __NL__ \

   ^

C:\Users\mikpal\Documents\Arduino\hardware\keyboardio\avr\libraries\Kaleidoscope\src/kaleidoscope/Kaleidoscope.h:203:40: note: in expansion of macro '_KALEIDOSCOPE_INIT_PLUGINS'

 #define KALEIDOSCOPE_INIT_PLUGINS(...) _KALEIDOSCOPE_INIT_PLUGINS(__VA_ARGS__)

                                        ^

C:\Users\mikpal\Documents\Arduino\libraries\Model01-Firmware-modded\Model01-Firmware-1.22-modded\Model01-Firmware-1.22-modded.ino:433:1: note: in expansion of macro 'KALEIDOSCOPE_INIT_PLUGINS'

 KALEIDOSCOPE_INIT_PLUGINS(

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

exit status 1
Error compiling for board Keyboardio Model 01.
```

I'm not sure what I should do more.

Help please?

obra, 2019-09-18:
Hi,

Can you confirm that the "latest" version of the Arduino IDE you're running
is 1.8.9?

Can you try version 1.96 of the Keyboardio hardware support package? I
suspect your custom firmware is using the current version of
Model01-Firmware, which pairs with a much newer Kaleidoscope than 1.22.

Best,
Jesse
ᐧ

mpalmujoki, 2019-09-23:
Hi Jesse,

Yes, I am using Arduino 1.8.9 (Windows Store 1.8.21.0).

Sure, I can try that. However I can't find the 1.96, or any version, of the "Keyboardio hardware support package". 

Can you please instruct me on where I can find the Keyboardio hardware support package? 

Best regards,
Mikael

obra, 2019-09-23:
Oh! The Windows Store builds of Arduino are known to be weirdly broken. Can
you give this tutorial a try?
https://github.com/keyboardio/Kaleidoscope/wiki/Install-arduino-support-on-Windows

Best,
Jesse
ᐧ

mpalmujoki, 2019-10-02:
Thanks, that helped!

closed 2019-10-02 by mpalmujoki