keyboardio/Model01-Firmware #91: when trying to flash keyboardio using the latest firmware and libraries results in IOCreatePlugInInterfaceForService failed

rreinhardt9, 2019-10-27:
Hello!

I'm on mac Catalina 10.15. I'm running into the following issue when trying to flash using the latest HEAD of master for the firmware with the latest libraries.

I have some local modifications... so to be sure those were not a factor I did a `git reset --hard` to the HEAD of this firmware repo and also ran the `make update-submodules` command in the `hardware/keyboardio` directory.

The only real change I've made since the last successful update was to update the libraries... although I wish I had tried to flash right before that because I can't remember for sure if I've flashed it since the upgrade to Catalina on mac. I'm a little concerned that maybe Catalina changes some permissions that are preventing access to the keyboard, but I'm not sure how to test that theory. If others are able to build successfully not on Catalina though that might be a good indicator 😆 

When starting the flash... sometimes the `prog` key turns red and sometimes it doesn't, but I get the same error message in both cases.

I'm not sure what else to try currently, has anyone else run into this issue or have any insights into where next to look for a solution?

```
12:14:05 ~/Documents/Arduino/Model01-Firmware(master) 
$ make flash
BOARD_HARDWARE_PATH="/Users/ross/Documents/Arduino/hardware" /Users/ross/Documents/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder flash
Building ./Model01-Firmware 0.0.0-g
v1.22-67-g4299 into /var/folders/lr/jfsdjryx2vbc0bdzt9qf2kl80000gn/T//kaleidoscope-ross/sketch/774572-Model01-Firmware.ino/output...
- Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0-g
v1.22-67-g4299.elf
  - Program:   25010 bytes (87.2% Full)
  - Data:       1338 bytes (52.3% Full)

To update your keyboard's firmware, hold down the 'Prog' key on your keyboard,
and then press 'Enter'.

When the 'Prog' key glows red, you can release it.

2019-10-27 12:16:09.176 system_profiler[38849:845882] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2019-10-27 12:16:12.428 system_profiler[38862:845975] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
Connecting to programmer: .
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = "
.
"; type = 

.   Software Version = 
; Hardware Version = ..
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
Connecting to programmer: .
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = "
.
"; type = 

.   Software Version = 
; Hardware Version = ..
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
make: *** [flash] Error 1
```

obra, 2019-10-27:
It’s a known bug in Catalina. We’re waiting on a fix from Apple.

rreinhardt9, 2019-10-27:
Awesome!!! 😌 I just got access to a laptop running Mojave 10.14.6 and verified I was still able to successfully make flash on there. Thanks for that info! 🙏  I'll go ahead and close this then since there is nothing for us to do here.

closed 2019-10-27 by rreinhardt9