keyboardio/Model01-Firmware #94: build error in LED-Stalker

faxm0dem, 2019-12-18:
Using a fresh clone I can't build the fw:

```
BOARD_HARDWARE_PATH="/home/fwernli/Arduino/hardware" /home/fwernli/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder build-all
Building ./Model01-Firmware 0.0.0-gv1.22-69-g2f9d into /tmp/kaleidoscope-fwernli/sketch/9138835-Model01-Firmware.ino/output...
In file included from /home/fwernli/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/Kaleidoscope-LED-Stalker.h:20:0,
                 from /home/fwernli/git/Model01-Firmware/Model01-Firmware.ino:55:
/home/fwernli/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/plugin/LED-Stalker.h:64:44: error: size of array 'map_' is not an integral constant-expression
     uint8_t map_[Runtime.device().numKeys()];
                                            ^
exit status 1
make: *** [/home/fwernli/Arduino/hardware/keyboardio/build-tools/makefiles//rules.mk:80: build-all] Error 1
```

faxm0dem, 2019-12-18:
If I remove the Stalker Plugin, I get this:
```
BOARD_HARDWARE_PATH="/home/fwernli/Arduino/hardware" /home/fwernli/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder build-all
Building ./Model01-Firmware 0.0.0-gv1.22-69-g2f9d into /tmp/kaleidoscope-fwernli/sketch/9138835-Model01-Firmware.ino/output...
In file included from /home/fwernli/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/Kaleidoscope-LED-Stalker.h:20:0,
                 from /home/fwernli/git/Model01-Firmware/Model01-Firmware.ino:55:
/home/fwernli/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/plugin/LED-Stalker.h:64:44: error: size of array 'map_' is not an integral constant-expression
     uint8_t map_[Runtime.device().numKeys()];
                                            ^
exit status 1
make: *** [/home/fwernli/Arduino/hardware/keyboardio/build-tools/makefiles//rules.mk:80: build-all] Error 1
```

noseglasses, 2019-12-18:
@faxm0dem, which Arduino version are you using?

noseglasses, 2019-12-18:
And which OS?

faxm0dem, 2019-12-18:
I didn't know the `make` command used arduino behind the scenes.
I have multiple versions installed, and the "current" one in my linux (debian buster) `$PATH` is 1.8.10

noseglasses, 2019-12-18:
Could you please try one of the older versions like v1.8.8 or v1.8.9? The compiler version recently jumped from 5.x to 7.x which may have caused that problem.

faxm0dem, 2019-12-18:
but where and which arduino binary is called?

noseglasses, 2019-12-18:
There's an environment variable that is read by the build system. Either `ARDUINO_SDK_PATH` or `ARDUINO_PATH`. Can't check right now. Just see what `env` tells you.
It's supposed to point to your Arduino installation and can be used to switch between different versions.

noseglasses, 2019-12-18:
I forgot to mention that Kaleidoscope's build system uses CCACHE unter the hood. That's why when switching Arduino-versions, you need to remove the `/tmp/kaleidoscope-fwernli/ccache` directory (it will be re-created automatically).

noseglasses, 2019-12-18:
Just checked it, the correct env var is `ARDUINO_PATH`.

noseglasses, 2019-12-18:
@faxm0dem, I just tried to reproduce this on my Ubuntu 19.10 system with no success.

Could you report what `which avr-gcc` says on you system and what version `avr-gcc -v` does report. Do you maybe have a different version of the avr-gcc compiler installed as a Debian package?

algernon, 2019-12-18:
You can also `export VERBOSE=1` before running `make`, and then `kaleidoscope-builder` will run in verbose mode, and will tell you exactly which path it's invoking avr-gcc with.

noseglasses, 2019-12-18:
@algernon, that path will be `/tmp/kaleidoscope-fwernli/ccache/bin/avr-gcc` which is a symbolic link to ccache unfortunately. That's what complicates things a bit. ccache will look for avr-gcc and avr-g++ in `$PATH`.

faxm0dem, 2019-12-18:
I don't have avr-gcc installed... this is starting to seriously look like a PBCAK

faxm0dem, 2019-12-18:
okay so the verbose says:

```
/data/fwernli/tmp/arduino-1.8.5/hardware/tools/avr/bin/avr-g++
```

getting there :)

noseglasses, 2019-12-18:
Good. That shows that you do not have ccache installed and pointing `ARDUINO_PATH` to Arduino 1.8.8 is likely to fix your issue.

faxm0dem, 2019-12-18:
I know why, this was set in `~/.kaleidoscope-builder.conf`

faxm0dem, 2019-12-18:
works with 1.8.9
thanks so much for your quick help

closed 2019-12-18 by faxm0dem

noseglasses, 2019-12-18:
@algernon, what is our currently supported minimal Arduino version?

algernon, 2019-12-18:
1.6.7 is the minimum set in kaleidoscope-builder, but we really should be using 1.8.x, where I'd think x >= 8 would be good.