keyboardio/Model01-Firmware #97: ps/2 fallback support

rudolph9, 2020-02-05:
I'm trying to connect the model01 via a usb->ps/2 adapter but the looking in the bios the Port is not recognizing the device and listed as empty.

I don't actually have a standard ps/2 keyboard to verify the port is working correctly or another device to verify the adapter is working correctly but the Model 01 powers on and is responsive when typing so I suspect the model01 just doesn't have support for it?

The best explanation I can find is a random post on reddit https://www.reddit.com/r/buildapc/comments/bbpytn/b450_tomahawk_ps2_port_help/

Any background info or direction for enabling support would be greatly appreciated!

obra, 2020-02-05:
You want to make sure you're in boot protocol mode for starters - That puts
the Model 01 into a mode that's going to be easy for a cheap PS/2 converter
to understand. That said, I don't know that anyone's ever actually made
this kind of flow go.

Is this to connect a Model 01 to a computer that doesn't have USB?

It's definitely worth making sure this works with a regular USB keyboard
before you .spend a lot of time on making this go with a Model 01.

-j
ᐧ

hoijui, 2022-05-24:
For an other USB keyboard, to make this work, I had to use a .. somewhat special adaptor, that does something to ensure enough power is supplied to the keyboard. maybe it contains capacitors or .. I am not good in electronics, but there are adapters especially made for that, which cost a few bucks more, but also not much.