keyboardio/Model01-Firmware #101: Cannot buld with Kaleidoscope==07f3ca and firmware==03bee4

MaxG87, 2020-04-18:
I tried to update the codebase because I wanted to update my Model01-Firmware and to start hacking again. It appears to me that the firmware cannot be build with the recent version of Kaleidoscope.

I am on Debian Testing. I updated to Arduino 1.8.12 by `sudo rm -r /usr/local/arduino` first and then following the instructions in the wiki. I updated the code base by `rm -rf ~/Arduino` first and then following the instructions in the README.md.

When I change to the Model01-Firmware folder and run `make`, the following error message is produced:

```
[...]
In file included from /home/maxg/Arduino/hardware/keyboardio/avr/libraries/Model01-Firmware/Model01-Firmware.ino:17:0:
/home/maxg/.arduino15/packages/keyboardio/hardware/avr/1.22.0/libraries/Kaleidoscope/src/Kaleidoscope.h: In member function 'void Kaleidoscope_::setup(byte)':
/home/maxg/.arduino15/packages/keyboardio/hardware/avr/1.22.0/libraries/Kaleidoscope/src/Kaleidoscope.h:67:25: warning: unused parameter 'keymap_count' [-Wunused-parameter]
   void setup(const byte keymap_count)
                         ^~~~~~~~~~~~
At global scope:
cc1plus: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
cc1plus: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
In file included from /home/maxg/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/Runtime.h:19:0,
                 from /home/maxg/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope/plugin/FocusSerial.h:20,
                 from /home/maxg/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/Kaleidoscope-FocusSerial.h:20,
                 from /home/maxg/Arduino/hardware/keyboardio/avr/libraries/Model01-Firmware/Model01-Firmware.ino:24:
/home/maxg/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/src/kaleidoscope_internal/device.h:22:22: error: 'Device' in namespace 'kaleidoscope' does not name a type
 extern kaleidoscope::Device device;

[...]

               ^
In file included from /home/maxg/.arduino15/packages/keyboardio/hardware/avr/1.22.0/libraries/Kaleidoscope-EEPROM-Settings/src/Kaleidoscope/EEPROM-Settings.h:22:0,
                 from /home/maxg/.arduino15/packages/keyboardio/hardware/avr/1.22.0/libraries/Kaleidoscope-EEPROM-Settings/src/Kaleidoscope-EEPROM-Settings.h:21,
                 from /home/maxg/Arduino/hardware/keyboardio/avr/libraries/Model01-Firmware/Model01-Firmware.ino:20:
/home/maxg/.arduino15/packages/arduino/hardware/avr/1.6.21/libraries/EEPROM/src/EEPROM.h: At global scope:
/home/maxg/.arduino15/packages/arduino/hardware/avr/1.6.21/libraries/EEPROM/src/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^~~~~~
exit status 1
make: *** [/home/maxg/Arduino/hardware/keyboardio/build-tools/makefiles//rules.mk:80: build-all] Fehler 1
```

I am including only the first and the last error message due to size restrictions.

Is this a bug or is the error on my side?

MaxG87, 2020-04-19:
I tried to reproduce on my other machine which runs Ubuntu 19.10. There compilation works flawlessly:

```
$ make -j4
BOARD_HARDWARE_PATH="~/Arduino/hardware" ~/Arduino/hardware/keyboardio/avr/libraries/Kaleidoscope/bin//kaleidoscope-builder build-all
Building ./Model01-Firmware 0.0.0 into /tmp/kaleidoscope-.../sketch/30676835-Model01-Firmware.ino/output
- Size: firmware/Model01-Firmware/Model01-Firmware-0.0.0.elf
  - Program:   25506 bytes (89.0% Full)
  - Data:       1369 bytes (53.5% Full)
```

Is there something obvious I could do to investigate the issue further? Any apparent reason why it should not work on Debian Testing?

obra, 2020-04-20:
I suspect you had an older version of Kaleidoscope automatically downloaded
on the Debian box.

hoijui, 2022-05-24:
This can be closed, I guess

closed 2022-05-24 by algernon