keyboardio/Model01-Firmware #102: Add commit SHA to BUILD_INFORMATION string

genehack, 2020-05-03:
Now that I'm using and flashing multiple Model01s, it occurs to be it would be nice if the string that's output by the `MACRO_VERSION_INFO` included something to provide a clue about which firmware build a board is running -- like a build date, or, ideally, the Git SHA of the `HEAD` at build time. 

I looked into adding this, but it turned into a deep hole that ended up in `kaleidoscope-builder` and that's a big bite to chew. 

Is there an easy way to modify `BUILD_INFORMATION` dynamically, or is this something that would make sense to add elsewhere? I'm happy to send in a patch but I honestly have no idea where to even start with this.

obra, 2020-05-03:
That was always the -intent- "LOCALLY_BUILT" was kind of a cop-out.

The thing you need to do is to get -DBUILD_INFORMATION passed in to the
sketch.

Something like "make build LOCAL_CFLAGS='-DBUILD_INFORMATION=blah'"

kaleidoscope-builder is the tool that might want to grow a feature to pass
in git information for what you're building.

If that's not enough pointer, just let me know.

-j

genehack, 2020-05-06:
The `LOCAL_CFLAGS` bit solves my immediate problem. I may take a crack at solving the more general `kaleidoscope-builder` case, but ...there's a lot in that file to understand. 

(For anybody else wondering, my specific immediate solution looks like 
```
LOCAL_CFLAGS="-DBUILD_INFORMATION=\"commit:`git rev-parse --short HEAD`\"" make build flash
```

closed 2024-05-06 by genehack