keyboardio/Model01-Firmware #104: Error building after upgrading to Catalina: bad CPU type in executable

fredcallaway, 2020-05-31:


obra, 2020-05-31:
Hi Fred,

What version of the Arduino IDE are you using?

fredcallaway, 2020-05-31:
Wow, quick response! After updating Arduino with `brew`, the GUI says 1.8.12 (can't say what it was before though—or if the update actually affected the executable in /Applications). However, I am now able to "Verify" a fresh clone of this repo with Arduino.

Unfortunately, I can't compile my previous Model01-Firmware.ino file because `Kaleidoscope-HostPowerManagement.h` is missing, will try to take stuff out until it compiles successfully.

Another issue is that `make` still doesn't work for me (with the fresh copy that *does* work in the IDE). It produces an very long (13k lines) error message which you can find [here](https://gist.github.com/fredcallaway/a36ffeec309ac50e281a6b65e8701ba4).

fredcallaway, 2020-05-31:
It seems that several of the plugins I was using (e.g. https://github.com/keyboardio/Kaleidoscope-Qukeys) are no longer available?

fredcallaway, 2020-05-31:
I tried copying copying my back up of `/Documents/Arduino/hardware/keyboardio/avr/libraries` into the fresh install and now I'm getting errors compiling in the IDE:

```
Arduino: 1.8.12 (Mac OS X), Board: "Keyboardio Model 01"

Multiple libraries were found for "HID.h"
Model01-Firmware:67:10: fatal error: Kaleidoscope-HardwareTestMode.h: No such file or directory
 Used: /Users/fred/Library/Arduino15/packages/keyboardio/hardware/avr/1.22.0/libraries/HID
 #include "Kaleidoscope-HardwareTestMode.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
 Not used: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/HID
exit status 1
Kaleidoscope-HardwareTestMode.h: No such file or directory
```

I'm going to leave this be until I get more guidance. It's not super critical because I can still use my keyboard (just can't change any of the settings)

gedankenexperimenter, 2020-05-31:
> It seems that several of the plugins I was using (e.g. https://github.com/keyboardio/Kaleidoscope-Qukeys) are no longer available?

That plugin has been incorporated into the main Kaleidoscope repository, along with many others. You should be able to see that the standalone repository has been archived.

fredcallaway, 2020-05-31:
@gedankenexperimenter OK, so in the version that I initially downloaded that were separate libraries alongside Kaledoscope, but are now files within Kaleidoscope. Thanks!

@obra I was able to address the "Multiple libraries" issue by uninstalling Arduino (critically, `rm -r ~/Library/Arduino15`). I am now again able to compile the default .ino file, and am slowly and steadily adding back pieces of my previous configuration. On top of that, `make` works! 🎉

*Solution for future reference*: Remove and reinstall /Applications/Arduino.app (from the website) AND remove ~/Library/Arduino15

closed 2020-05-31 by fredcallaway