keyboardio/Model01-Firmware #111: variable or field 'anyKeyMacro' declared void

tsira, 2021-12-16:
Using Fresh Install, following https://kaleidoscope.readthedocs.io/en/latest/setup_toolchain.html#Arduino-Windows
OS: Win10, 
Arduino IDE: 1.8.18,
Boards: Kaleidoscope keyboards v 1.99.3

Using the latest master `05854a8` from: https://github.com/keyboardio/Model01-Firmware/blob/master/Model01-Firmware.ino

Getting the following error during Verify/Compile in the Arduino IDE:

Arduino: 1.8.18 (Windows 10), Board: "Keyboardio Model 01"
```

Model01-Firmware:303:25: error: variable or field 'anyKeyMacro' declared void

 static void anyKeyMacro(KeyEvent &event) {

                         ^~~~~~~~

Model01-Firmware:303:25: error: 'KeyEvent' was not declared in this scope

Projects\Model01-Firmware\Model01-Firmware.ino:303:25: note: suggested alternative: 'Key_Menu'

 static void anyKeyMacro(KeyEvent &event) {

                         ^~~~~~~~

                         Key_Menu

Model01-Firmware:303:35: error: 'event' was not declared in this scope

 static void anyKeyMacro(KeyEvent &event) {

                                   ^~~~~

Model01-Firmware:320:46: error: 'KeyEvent' has not been declared

 const macro_t *macroAction(uint8_t macro_id, KeyEvent &event) {

                                              ^~~~~~~~

Projects\Model01-Firmware\Model01-Firmware.ino: In function 'const macro_t* macroAction(uint8_t, int&)':

Model01-Firmware:324:28: error: request for member 'state' in 'event', which is of non-class type 'int'

     versionInfoMacro(event.state);

                            ^~~~~

Model01-Firmware:328:5: error: 'anyKeyMacro' was not declared in this scope

     anyKeyMacro(event);

     ^~~~~~~~~~~

exit status 1

variable or field 'anyKeyMacro' declared void

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

obra, 2021-12-16:
Do things work if you use the version of Model01-Firmware included with the Kaleidoscope bundle?

tsira, 2021-12-16:
Using: https://github.com/keyboardio/Kaleidoscope/blob/master/examples/Devices/Keyboardio/Model01/Model01.ino

different error: ` 'KeyEvent' has not been declared`

are there additional files aside from that .ino needed for compilation?

```
Model01-Firmware:88:46: error: 'KeyEvent' has not been declared
 const macro_t *macroAction(uint8_t macro_id, KeyEvent &event) {
                                              ^~~~~~~~
In file included from Arduino15\packages\keyboardio\hardware\avr\1.99.3\libraries\Kaleidoscope\src/kaleidoscope/key_events.h:23:0,
                 from Arduino15\packages\keyboardio\hardware\avr\1.99.3\libraries\Kaleidoscope\src/Kaleidoscope.h:83,
                 from Model01-Firmware\Model01-Firmware.ino:20:
Model01-Firmware\Model01-Firmware.ino: In function 'const macro_t* macroAction(uint8_t, int&)':
Model01-Firmware:89:43: error: request for member 'state' in 'event', which is of non-class type 'int'
   if (macro_id == 1 && keyToggledOn(event.state)) {
                                           ^
Arduino15\packages\keyboardio\hardware\avr\1.99.3\libraries\Kaleidoscope\src/kaleidoscope/keyswitch_state.h:34:35: note: in definition of macro 'keyIsPressed'
 #define keyIsPressed(keyState)  ((keyState) & IS_PRESSED)
                                   ^~~~~~~~
Model01-Firmware\Model01-Firmware.ino:89:24: note: in expansion of macro 'keyToggledOn'
   if (macro_id == 1 && keyToggledOn(event.state)) {
                        ^~~~~~~~~~~~
Model01-Firmware:89:43: error: request for member 'state' in 'event', which is of non-class type 'int'
   if (macro_id == 1 && keyToggledOn(event.state)) {
                                           ^
Arduino15\packages\keyboardio\hardware\avr\1.99.3\libraries\Kaleidoscope\src/kaleidoscope/keyswitch_state.h:39:35: note: in definition of macro 'keyWasPressed'
 #define keyWasPressed(keyState) ((keyState) & WAS_PRESSED)
                                   ^~~~~~~~
Model01-Firmware\Model01-Firmware.ino:89:24: note: in expansion of macro 'keyToggledOn'
   if (macro_id == 1 && keyToggledOn(event.state)) {
                        ^~~~~~~~~~~~

```

gedankenexperimenter, 2021-12-17:
Kaleidoscope v1.99.3 is now almost a year old, and doesn't have `KeyEvent` in it.  The latest `master` branch of Model01-Firmware should be build with the latest Kaleidoscope `master`, not with v1.99.3.

tsira, 2021-12-18:
Thanks! Was able to find the correct `.json` file to  add to the additional boards manager urls to update my Kaleidoscope Board version. 
```https://raw.githubusercontent.com/keyboardio/arduino-kaleidoscope-master/main/package_kaleidoscope_master_index.json```

https://community.keyboard.io/t/macros-not-working-with-v1-99-3-for-atreus-and-java-lang-nullpointerexception-when-trying-to-get-the-master-branch-installed-in-the-arduino-ide/5010/8

closed 2021-12-18 by tsira