keyboardio/Model01-Firmware #112: Relationship with kaleidoscope repo

jan-warchol, 2022-02-22:
Context: I need to make an update to my Model01 firmware (after forking this repo a long time ago and making some changes in the layout).

I'm having trouble understanding the relationship between this repo and Kaleidoscope. In particular, I found https://github.com/keyboardio/Kaleidoscope/tree/master/examples/Devices/Keyboardio/Model01, which seems to be Model01 firmware too, but the `.ino` files are significantly different. Which one should I use? I'm fine with starting from scratch and re-doing my layout on top of current default firmware, but I'd like to know what exactly should I use as a starting point. Currently I'm quite confused.

algernon, 2022-02-22:
This repository is the official default firmware for the Keyboardio Model01. The one in the Kaleidoscope is merely a basic example.

I'd suggest continuing to use this repo as the base of your own modifications.

jan-warchol, 2022-02-22:
wow, that was fast!

Okay. I suggest adding some note, perhaps a short README, to the Kaleidoscope example to make this clearer.
BTW, I think that part of my confusion stemmed from the fact that this repo readme instructs users to install & configure arduino using these instructions from kaleidoscope: https://kaleidoscope.readthedocs.io/en/latest/setup_toolchain.html, and it's not fully clear which part is the common setup and which corresponds to using GUI. I wish the instructions were clearer.

PS I've read your blog, good stuff :)

jan-warchol, 2022-02-22:
PPS anyway, thank you! :)

hoijui, 2022-05-24:
I had the same confusion, and actually used the example version from kaleidoscope, which caused me problems (as it is limited, e.g. no Crysalis support), which @algernon helped me overcome in chat (thanks again!)
I would preffer if the kaleidoscope example would be removed alltogether, and just be replaced with a link (or inclusion of this repo as a git submodule).

algernon, 2022-05-24:
I'd rather go the other way around: update the example in Kaleidoscope, and declare that the default firmware. We do that for the Keyboardio Atreus and the Model 100, we could do it for the Model01 too.

The reason it wasn't done before is - I believe - historical: with a separate repository, it was easier to iterate on both independently. We no longer need to do that, so may aswell turn _this_ repo into a link.

What do you think, @obra?

obra, 2022-05-24:
Indeed, I think we probably ought to be moving the default firmware into
the Kaleidoscope repo. We may want to think about having it live in th
examples dir for the hardware plugin if the IDE will detect that.

algernon, 2022-05-24:
I don't think the IDE detects examples in any other place than the top-level `examples` dir. We _could_ still have it in the hw plugin dir, and just copy it over during package build, like we juggle the `plugins/` dir around too.

obra, 2022-05-24:
it should detect the examples for any arduino library, which our plugins are.

algernon, 2022-05-24:
Good point. I'll give it a try.

algernon, 2022-06-06:
> it should detect the examples for any arduino library, which our plugins are.

I just tried this, and indeed, it works, so we can move the factory firmware files to the plugin dirs.