keyboardio/Model01-Firmware #114: How to have multiple "primary" layers?

hoijui, 2022-05-24:
... as in, I would want Colemak as my primary, and QWERTY on a different layer.
I know how to do it with c&p, but I wondered if it would make sense to restructure the code, so that the primary layers all get defined, but not as `[PRIMARY]`, but as '[QWERT]', `[COLEMAK]` and so on, and then by default, one of those gets assigned to `[PRIMARY]`, and the one of the others could be used as an additional layer.
... or is that not possible because the layers then need to be adjusted anyway to add the function keys to change between them?

algernon, 2022-05-24:
The reason we haven't done that is that any additional layer is +128 bytes of PROGMEM, and we're kinda short on that. For practical reasons, we also need at least the same amount of layers in EEPROM, which we're even more short on. There are a few other considerations too, having to renumber the layers being one of them, indeed.

If you want multiple primary layers, I'd suggest forking the repository, and making your own modification: replace the [layer list](https://github.com/keyboardio/Model01-Firmware/blob/9cc38674c04287abb94771a923f96a23eda4bcc0/Model01-Firmware.ino#L145) with: `enum { QWERTY, COLEMAK, NUMPAD, FUNCTION };`, remove the `#if`s around QWERTY and Colemak, and use `QWERTY` / `COLEMAK` respectively, instead of `PRIMARY`, and you're pretty much good to go.

If need be, I can edit it for you and show a full example, rather than just the above description.

hoijui, 2022-05-25:
Thank you! :-)
that i certainly good enough for me as an explanation, and hopefully for others too.
I can do it myself the way you explained; thank you for the kind offer!

closed 2022-05-25 by hoijui